HumanoidPequi/HumanoidPMec_LARC2023 #7: Rosserial

lucasolives, 2023-05-20:


lucasolives, 2023-05-20:
1. Open up terminal and type: `roscore`  
2. In a new terminal type: `rosrun rosserial_arduino serial_node.py _port:=/dev/ttyUSB0`
3. Type `rostopic list` to make sure that everything is running

lucasolives, 2023-05-20:
**[ERROR] [1684614705.969329]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino**

lucasolives, 2023-05-20:
Possible solution: 
If rebuilding isn't an option you can add the definitions manually. On the arduino side you'll have a ros_lib folder. You have two sub directories you need to change.

The first is ./ros/node_handle.h. At the top of the file you'll see a few protocol versions defined as const uint8_t. Add in the version id for noetic as const uint8_t PROTOCOL_VER6 = 0xfa; then change the line that states what version should be used. const uint8_t PROTOCOL_VER = PROTOCOL_VER6;.

Lastly, you just need to make this same change in ./ros_lib/ros/node_handle.h

lucasolives, 2023-05-21:
The code **controle_baixo_nível.ino** is being developed, so it has some things that need to be fixed. 1 - `servo_cb()` is able to receive only one angle when it should get two. 
Possible solution: Create a `custom_msg` package that contains two UInt16 values.

andrecaraiba, 2023-06-01:
This tutorial solved the problem: https://answers.ros.org/question/394805/esp32-keep-rebooting-with-rosserial/

PR #17 (2023-06-09): Esp
closes #7 
closes #8 
closes #13 
closes #10
closed 2023-06-09 by dimitriasilveria