HumanoidPequi/HumanoidPMec_LARC2023 #8: Trouble with Dynamixel signal

alvinolima50, 2023-05-29:


alvinolima50, 2023-05-29:
What can it be?
The cable impedance is different from the dynamixel input impedance, causing a reflection of the signal interfering with the input signal.

Options to solve the problem: avoid this reflection!

- [ ] level shifter position change
- [ ] Add Diode 5V
- [ ]

andrecaraiba, 2023-06-01:
Possible cause: Incompatibility between the Dynamixel firmware and the Dynamixel2Arduino library. To resolve the issue, it was necessary to change the firmware version from 45 to 40. You can use this reference to update the firmware version by following the steps outlined in Wizard 2.0: [Update firmware version](https://m.youtube.com/watch?v=aVZytXRc_r8).

Additionally, it has been discovered that the signal reflection is actually a return packet containing status information that the Dynamixel returns. You can find more details about this in the [Dynamixel Protocol 1.0 documentation](https://emanual.robotis.com/docs/en/dxl/protocol1/).

Furthermore, for the Dynamixel to function properly, the microcontroller needs to receive these return packet information. This requires a communication circuit between the Dynamixel and the microcontroller. You can refer to the [MX-28 documentation documentation](https://emanual.robotis.com/docs/en/dxl/mx/mx-28/), the circuit is also located in the hardware folder of the repository.

PR #17 (2023-06-09): Esp
closes #7 
closes #8 
closes #13 
closes #10
closed 2023-06-09 by dimitriasilveria